linuxwacom/wacom-hid-descriptors #182: Intuos2 (9x12)

slalomsk8er, 2021-12-26:
I think it works great besides the mouse wheel isn't working in my desktop yet. It sends `Event: time 1640559877.689042, type 3 (EV_ABS), code 6 (ABS_THROTTLE), value 856` events so there seems to be something missing somewhere and I'm on the search. This is how I found this repo and it looks like you are missing the descriptors for this old piece of hardware so here it is
[sysinfo.ZPgBtGyXBh.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7777736/sysinfo.ZPgBtGyXBh.tar.gz)
.

jigpu, 2022-01-07:
Thanks for the sysinfo -- I've pushed it up to the repository. It looks like the wheel is being reported as an ABS_THROTTLE event, which appears to be correct. Unfortunately there is no scrollwheel emulation for this control -- it can only be used as an additional absolute axis inside of programs like GIMP or Krita. In theory such emulation could be added in both xf86-input-wacom and libwacom, but its unlikely any developers are going to work on such a feature since the 4D mouse is such an old device...

closed 2022-01-07 by jigpu

slalomsk8er, 2022-01-07:
Thanks, I documented my workaround here https://github.com/linuxwacom/xf86-input-wacom/issues/204#issuecomment-1001776803