linuxwacom/wacom-hid-descriptors #185: HP ProBook X360 11 G5 EE

MSmolarz, 2022-01-01:
Hi,
On my HP ProBook X360 11 G5 EE the stylus is not recognized correctly (in GNOME settings), but it works.
The tarball generated by the [sysinfo script](https://github.com/linuxwacom/wacom-hid-descriptors/files/7798099/sysinfo.B2HOmo20Hc.tar.gz) is attached.
Is it possible that I can pre-install the configuration on my laptop before it is delivered via the official update?
If so, how should I proceed?
Thanks
Mark

MSmolarz, 2022-01-02:
I've tried to make a .tablet file by myself, because the automatic generation by the [git.update skript](https://github.com
/linuxwacom/wacom-hid-descriptors/blob/master/README.md#maintainers) doesn't work because of a missing "hidrd-convert" package.
This is my result: [elan1001.zip](https://github.com/linuxwacom/wacom-hid-descriptors/files/7799510/elan1001.zip)
Is this correct?

jigpu, 2022-01-07:
Thanks for the sysinfo -- I've pushed it up to the repository.

As for the tablet definition, there are a few small changes I'd make:
* Rename to elan-2a70.tablet
* Include a commented link to this issue at the top of the file (e.g. `# https://github.com/linuxwacom/wacom-hid-descriptors/issues/185`)
* Width and height should be rounded to the nearest whole number
* IntegratedIn should be "Display;System"

After you make those changes I would open a [libwacom issue](https://github.com/linuxwacom/libwacom/issues) with your tablet definition for them to integrate :)

closed 2022-01-07 by jigpu

MSmolarz, 2022-01-10:
Hello,
thank you for the feedback.
I have made the corrections you suggested.
[elan-2a70.tablet.zip](https://github.com/linuxwacom/wacom-hid-descriptors/files/7840731/elan-2a70.tablet.zip)

jigpu, 2022-01-11:
Looks good! I would recommend filing the libwacom issue yourself in case there are any other questions they have for you.