linuxwacom/wacom-hid-descriptors #196: PINE64 PineNote

utcafj, 2022-01-30:
[sysinfo.n2dS2k0dtJ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7966091/sysinfo.n2dS2k0dtJ.tar.gz)

utcafj, 2022-01-30:
Another one with my custom pinenote.tablet file removed: 
[sysinfo.QLV8DTvIWZ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7966099/sysinfo.QLV8DTvIWZ.tar.gz)

jigpu, 2022-01-31:
Thanks for the sysinfo! I've gone ahead and opened a pull request at libwacom for the tablet definition. Note that the definition is a little different from the version included with your first sysinfo file (the `Styli=` line has been corrected with a proper set of EMR IDs).

closed 2022-01-31 by jigpu

utcafj, 2022-01-31:
Thanks for the quick response! What is the proper way to pair it with the touchscreen (named cyttsp5 in libinput.txt after the driver), which is connected via i2c?

utcafj, 2022-01-31:
I've attached the output of the attribute walk. This one does not provide a vendor or product attribute (yet):
[udevadm_cyttsp.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/7974906/udevadm_cyttsp.txt)

jigpu, 2022-02-08:
Both of the devices should already be using the libwacom driver, so if it is not providing the desired experience you can try moving them both over to the xf86-input-wacom driver instead. I'm not 100% sure if the `MatchProduct` line will work properly, but if it does then both the pen and touch should show up in the output of `xsetwacom`.

Save the following at `/etc/X11/xorg.conf.d/90-pinenote-pentouch.conf` and reboot:
~~~
# Force PINE64 PineNote touchscreen and pen devices to use the xf86-input-wacom driver.

Section "InputClass"
        Identifier "Pine64 touchscreen OVERRIDE"
        MatchProduct "cyttsp5"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection

Section "InputClass"
        Identifier "Pine64 stylus OVERRIDE"
        MatchUSBID "2d1f:0095"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
~~~