linuxwacom/wacom-hid-descriptors #197: Getac UX10 tablet not recognized

TidelyPom, 2022-02-02:
Hi

I have a Getac UX10 tablet that isn't recognised by Ubuntu 20.04, I have run the sysnsinfo.sh script and I am sending you the tarball as recommended .
[sysinfo.s4LlfzTr3g.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7987229/sysinfo.s4LlfzTr3g.tar.gz)

jigpu, 2022-02-02:
It looks like this device is using a touchscreen digitizer from eGalax / EETI, which our sysinfo script largely ignored. I've updated it to no longer skip over devices from EETI. Can you please run the [updated sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach its results?

Does your UX10 use the "digitizer pen" (https://www.getac.com/us/accessories/digitizer-pen-gmpdx6/) or the "capacitive pen" (https://www.getac.com/us/accessories/capacitive-hard-tip-stylus-tether-gmpsxl/)?

TidelyPom, 2022-02-03:
Thank you. The pen is the capacitive pen. Here is the new tarball.

[sysinfo.KtRbPRYHog.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/7994219/sysinfo.KtRbPRYHog.tar.gz)

jigpu, 2022-02-09:
Thanks! The update seems to have done the trick and dumped data for the eGalax / EETI sensor correctly. It looks like this particular model only uses a plain touchscreen sensor without any special digitizer handling. Because of this, it doesn't need an tablet definition file in libwacom -- there are no digitizer properties to be configured. I've added the sysinfo to the database for future reference, however, since it seems there is a sibling model which may have special digitizer handling...

The device shows up in the output of `xinput list` which indicates that it "should" be working. If it doesn't, you may need to contact the libinput developers (https://gitlab.freedesktop.org/libinput/libinput/) to diagnose why not. You can include a link to this issue since they would find the sysinfo output handy.

closed 2022-02-09 by jigpu