linuxwacom/wacom-hid-descriptors #199: HP Pavilion x360 hp hp 14-baxxx Touchscreen not detected

serhumanos, 2022-02-04:
Hi, What can I do to make my touchscreen be recognized by wacom utility? I was reading some docs ans answers and they said that I need descriptors because the touchscreen is ELAN. I only find for 14-cd0xxx [here](https://github.com/linuxwacom/wacom-hid-descriptors/tree/master/HP%20Pavilion%20x360%20Convertible%2014-cd0xxx) but I don't know how to load to test it if it works (I suppose it don't).
I appreciate your help. thanks!

jigpu, 2022-03-17:
Please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball to this issue (simply drag & drop the file onto the comment box). This will let us make sure that your sensor has the same ID as the one that you've linked. From there we can see what the next step will be.

serhumanos, 2022-03-17:
OK thanks!
here is the file
[sysinfo.j0LyeiD0LP.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/8294228/sysinfo.j0LyeiD0LP.tar.gz)

serhumanos, 2022-04-11:
https://github.com/linuxwacom/libwacom/issues/488

> Thanks! I've pushed the sysinfo up to the repository. It does indeed appear that your particular device uses an Elan sensor that is not currently recognized by libwacom. The libwacom library is what allows software like the GNOME Control Center to recognize tablet devices.
> 
> I can't help with creating the tablet definition for your device (my scripts here don't know how to process Elan sensors), but the libwacom developers will be able to help. As a next step, please [open a new libwacom issue](https://github.com/linuxwacom/libwacom/issues) requesting help creating a tablet definition file. Include a link back to this issue so they can have access to the data.

OK! it is done, the new issue is [here](https://github.com/linuxwacom/libwacom/issues/488).
thanks!