linuxwacom/wacom-hid-descriptors #203: Huion Kanvas 13

Ranguna, 2022-02-13:
Hello!

Seems gnome is not picking up my huionkanvas 13 (no the pro version). I've attached the tar ball here with the sys dump :)
[sysinfo.BFrex7lf87.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/8055545/sysinfo.BFrex7lf87.tar.gz)

jigpu, 2022-03-17:
Thanks for the sysinfo. I've pushed it up to the repository, but you'll need to file an issue at https://github.com/linuxwacom/libwacom/issues to request help in creating a tablet definition file. The Huion devices are a bit weird in that several devices all share a single USB product ID, which requires special handling by libwacom. The developers over there should be able to assist you in creating an appropriate file. Be sure to include a link back to this issue if the raw tablet data is needed for reference.

closed 2022-03-17 by jigpu

Ranguna, 2022-03-17:
Cool, thanks @jigpu !