linuxwacom/wacom-hid-descriptors #204: HP ProBook x360 435 G8

blk-ole, 2022-02-14:
Hi,

here is the data for the HP ProBook x360 435 G8, which is still missing from the repo.
The issue with stylus detection is, that the buttons are not properly recognized.
A press on the first button is recognized as lifting the tip and the second button is recognized as normal button press (left mouse).

The first button should be the left mouse button and the second button the right mouse button accordingly
[sysinfo.j64MOYmSS6.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/8059298/sysinfo.j64MOYmSS6.tar.gz)
.

jigpu, 2022-02-17:
Thanks for the sysinfo -- I've added it to the repository. It looks like this device uses an Elan sensor so we can't auto-generate a tablet definition file. You may want to request help from https://github.com/linuxwacom/libwacom/issues in creating a tablet definition file. You should include a link to this issue so they have the HW info they need to write one by hand.

It looks like the pen hardware is configured to have one of the buttons act as an eraser switch and the other act as a mouse button. Its possible the drivers powering your pen (libinput and hid-generic) are misbehaving. I would follow up with a new issue at https://gitlab.freedesktop.org/libinput/libinput/-/issues -- again with a link back here so they have the HW info.

blk-ole, 2022-06-08:
Related issue for libinput:
https://gitlab.freedesktop.org/libinput/libinput/-/issues/776