linuxwacom/wacom-hid-descriptors #214: Adding descriptors for Huion Q620M tablet

woj-o, 2022-03-31:
Hello.
Please find tarball from sysinfo.sh about that tablet.
It mostly works out of the box (stylus, keys), but it is recognized as H950p (same USB signature).
Due to that some features of it (wheel, tilt?) do not work.
I'm happy to help getting the needed data (events, etc), but would need help of how to gather them.

Cheers
Wojtek
[sysinfo.6Qz87YOL9o.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/8390849/sysinfo.6Qz87YOL9o.tar.gz)

jigpu, 2022-04-11:
Thanks for the sysinfo. I've pushed it up to the repository, but you'll want to [open a libwacom issue](https://github.com/linuxwacom/libwacom/issues) to have them help you create a tablet definition for the Q620M. I know they have a way to resolve the detection issue caused by same USB IDs.

You will probably also need to contact the developer of drivers for for your tablet (probably [Digimend](https://digimend.github.io/)?) to see if they can help you get the wheel/tilt features working.

closed 2022-04-11 by jigpu

woj-o, 2022-04-11:
Hello.

Same id is not a problem ( i can just use file for mine tablet and get rid of others ) - not plug and play but works:)

Yeah digimend was second option and v10 supports this tablet ( tilt and dial) but they need to be compiled. Not sure why from repo it wasn't working. 

So Now I'm on just digimend plus xwacom set-up ( to get KDE mapping tool) 

So I'm actually quite good with that combination.

If I can help somehow to get this model into libwacon I'm happy to help - not sure if that's possible though as wacom driver itself seems just to support basic stylus and buttons. 
Dial responds to events but as button ( press clockwise release counterclockwise) 

⁣Pobierz aplikację BlueMail dla systemu Android ​

***@***.***>

jigpu, 2022-04-13:
> If I can help somehow to get this model into libwacon I'm happy to help - not sure if that's possible though as wacom driver itself seems just to support basic stylus and buttons.
> Dial responds to events but as button ( press clockwise release counterclockwise)

Its absolutely possible to get this model into libwacom :) You'll need to [file a libwacom issue](https://github.com/linuxwacom/libwacom/issues/) with a link back here so that the necessary information is available. Getting it into libwacom won't improve the support for the stylus/buttons/dial since that's a driver issue, but it will potentially help other users in the future make better use of their device if the driver improves in the future.