linuxwacom/wacom-hid-descriptors #227: Lenovo Yoga 9 14IAP7

mmxgn, 2022-05-01:
Hi,

This is the Lenovo 9i 2022 edition. It doesn't seem to recognize the stylus/tablet on Ubuntu 22.04LTS (libwacom 2.2.0) when going to Ubuntu Settings->Wacom Panel. Although it seems to work (as a pointing device) the button closest to the tip doesn't seem to do anything and the second seems to work as a middle mouse button.

Here's my sysinfo:

[sysinfo.HH7xAqOQyE.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/8598546/sysinfo.HH7xAqOQyE.tar.gz)

RayJW, 2024-12-11:
I saw that there were some updates to the script for generating the tarball. I took the liberty and generated a fresh one just in case, that makes integration easier: [sysinfo.wCbd9PWIdo.tar.gz](https://github.com/user-attachments/files/18097916/sysinfo.wCbd9PWIdo.tar.gz)

closed 2025-01-08 by jigpu

jigpu, 2025-01-08:
Data has been merged. This device doesn't seem to have a Wacom sensor, so you may want to file an issue with libwacom for assistance getting your device supported.

commit f3684ae9 (2025-01-08): Add multiple sysinfo logs

* Lenovo Yoga 6 13ALC6
  sysinfo.mSCxAg3Y7r
  Closes #244

* Asus Q405UA-BI57T
  sysinfo.ROb2lagkqG
  Closes #243

* XP-Pen Deco L
  sysinfo.y9iVOhZiPM
  Closes #242

* Chuwi MiniBook X
  sysinfo.CSujtSF02A
  Closes #241

* GPD Pocket 3
  sysinfo.4p9auNwjkC
  Closes #240

* Dell Latitude 7210 2 in 1
  sysinfo.oN0r97Rlcn
  Closes #239

* XP-Pen Artist 12
  sysinfo.BPdqfy3fcP
  Closes #238

* Huion Kamvas Pro 13 (2.5K)
  sysinfo.m8z8elNbuM
  Closes #237

* Lenovo IdeaPad Flex 5 16ALC7
  sysinfo.ZBg72Fo2lq
  Closes #236

* Lenovo 20LHS04L00 ThinkPad X380 Yoga
  sysinfo. [... cut]