linuxwacom/wacom-hid-descriptors #231: Dell Latitude 5289 2in1 Wacom HID 4846 Pen&Touch

teknobilly, 2022-05-31:
Dell Latitude 5289 2in1 w/ Wacom HID 4846 Pen & Touch 
EndeavourOS

JournalCtl
event11 - Wacom HID 4846 Pen: tablet 'Wacom HID 4846 Pen' unknown to libwacom

sysinfo.sh running X.ORG
[sysinfo.1m89LEwy7P.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/8801753/sysinfo.1m89LEwy7P.tar.gz)

sysinfo.sh running WAYLAND (If it makes any difference)
[sysinfo.jzY3t7djf2.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/8801754/sysinfo.jzY3t7djf2.tar.gz)

:) Thank you for the magic you do

teknobilly, 2022-06-03:
Previous sysinfo was missing results from xsetwacom.   Re-ran with more better results.
[sysinfo.8FMoriDTcJ.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/8835473/sysinfo.8FMoriDTcJ.tar.gz)

jigpu, 2024-12-10:
Pull request merged.

closed 2024-12-10 by jigpu