linuxwacom/wacom-hid-descriptors #240: GPD Pocket 3 / GXTP7380:00

AcouBass, 2022-07-22:
Got this device to show up correctly in GNOME Settings panel using a homemade .tablet file but I've no idea if I did that fully and correctly so thought it'd be better to send the sysinfo file here and make sure everything's as good as it can be!

[sysinfo.4p9auNwjkC.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/9169527/sysinfo.4p9auNwjkC.tar.gz)

Thanks :)

michimussato, 2022-10-02:
Hi @AcouBass. Got the same hardware like you. Can you share your `.tablet` and `.stylus` files with me? I currently have the issue that the tilt direction of the pen (Microsoft Surface Pen) is incorrect. One of the two axis is applying inverted tilt values (tested in Krita, leading to incorrect strokes with tilt supported tools). I'm not sure yet which axis (`ABS_TILT_X` or `ABS_TILT_Y`). Are you familiar with the behavior? And does your own definition solve that problem?

AcouBass, 2022-10-02:
And great,

michimussato, 2022-10-02:
Wow, you're quick :) Thanks for sharing! 

So far, I've only tried to add a custom stylus definition (work in progress)"
```
[...]
[0x0]
# type 1 (Event EV_KEY)
#        - BTN_TOOL_PEN (code 320)  (proximity bool)
#        - BTN_TOOL_RUBBER (code 321)  (proximity bool)
#        - BTN_TOUCH (code 330)  (touching bool)
#        - BTN_STYLUS (code 331)  (side button bool)
#        (Property INPUT_PROP_DIRECT)
# type 1 (EV_KEY), code 320 (BTN_TOOL_PEN)    Bool	Proximity
# type 1 (EV_KEY), code 330 (BTN_TOUCH)       Bool      Touch (For tip and eraser)
# type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER) Bool	Proximity
# type 3 (Event EV_ABS)
#        - ABS_X (code 0)
#        - ABS_Y (code 1)
#        - ABS_PRESSURE (code 24) (for tip and eraser)
#        - ABS_TILT_X (code 26) (for tip only)
#        - ABS_TILT_Y (code 27) (for tip only)
# type 3 (EV_ABS), code 24 (ABS_PRESSURE)
# type 3 (EV_ABS), code 1 (ABS_Y)
# type 3 (EV_ABS), code 0 (ABS_X)
# type 4 (Event EV_MSC)
#        - MSC_SCAN (code 4)
#          - value d0042 (tip)
#          - value d0044 (side button)
#          - value d0045 (eraser)
# type 4 (EV_MSC), code 4 (MSC_SCAN)

Name=Microsoft Surface Pen Pen
Group=isdv4-aes
#PairedStylusIds=0xffffe;
Buttons=2
Axes=Tilt;Pressure;Distance;
Type=General
[...]
```

Without this definition, I got (have to disconnect my Wacom One S first):
```
[michael@pocket3 ~]$ sudo libwacom-show-stylus 
Using "GXTP7380:00 27C6:0113 Stylus": /dev/input/event9
Using stylus file(s): /usr/share/libwacom/libwacom.stylus
Tool id 0x0 serial 0x0 in-proximity: True 
^CTerminating
Unknown stylus id 0x0. New entry needed
Suggested line for .tablet file:
Styli=
```

Now I get:
```
[michael@pocket3 ~] $ sudo libwacom-show-stylus
Using "GXTP7380:00 27C6:0113 Stylus": /dev/input/event7
Using stylus file(s): /usr/share/libwacom/libwacom.stylus
Tool id 0x0 serial 0x0 in-proximity: True 
^CTerminating
Suggested line for .tablet file:
Styli=isdv4-aes
```
So, I can now define specific settings for that pen. However, the tilt recognition worked out of the box after setting `wacom` as the driver instead of `libinput` for Xorg.
```
[michael@pocket3 ~] $ cat /etc/X11/xorg.conf.d/99-touchscreen.conf 
# https://wiki.archlinux.org/title/GPD_Pocket_3#Touchscreen_and_Digitizer

Section "InputClass"
  Identifier    "calibration"
  Driver        "wacom"
[... 2588 lines cut ...]
Event: time 1664718791.656712, -------------- SYN_REPORT ------------
Event: time 1664718791.661449, type 3 (EV_ABS), code 0 (ABS_X), value 1251
Event: time 1664718791.661449, -------------- SYN_REPORT ------------
Event: time 1664718791.665533, type 3 (EV_ABS), code 0 (ABS_X), value 1272
Event: time 1664718791.665533, -------------- SYN_REPORT ------------
Event: time 1664718791.669118, type 3 (EV_ABS), code 0 (ABS_X), value 1290
Event: time 1664718791.669118, -------------- SYN_REPORT ------------
Event: time 1664718791.671820, type 3 (EV_ABS), code 0 (ABS_X), value 1306
Event: time 1664718791.671820, type 3 (EV_ABS), code 1 (ABS_Y), value 5341
Event: time 1664718791.671820, -------------- SYN_REPORT ------------
Event: time 1664718791.676618, type 3 (EV_ABS), code 0 (ABS_X), value 1315
Event: time 1664718791.676618, type 3 (EV_ABS), code 1 (ABS_Y), value 5339
Event: time 1664718791.676618, -------------- SYN_REPORT ------------
Event: time 1664718791.680530, type 3 (EV_ABS), code 0 (ABS_X), value 1320
Event: time 1664718791.680530, type 3 (EV_ABS), code 1 (ABS_Y), value 5335
Event: time 1664718791.680530, -------------- SYN_REPORT ------------
Event: time 1664718791.684102, type 3 (EV_ABS), code 0 (ABS_X), value 1309
Event: time 1664718791.684102, type 3 (EV_ABS), code 1 (ABS_Y), value 5330
Event: time 1664718791.684102, -------------- SYN_REPORT ------------
Event: time 1664718791.686901, type 3 (EV_ABS), code 0 (ABS_X), value 1285
Event: time 1664718791.686901, type 3 (EV_ABS), code 1 (ABS_Y), value 5322
Event: time 1664718791.686901, -------------- SYN_REPORT ------------
Event: time 1664718791.691537, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 0
Event: time 1664718791.691537, -------------- SYN_REPORT ------------
```

michimussato, 2022-10-02:
(I suspect that the solution to the tilt issue can be found somewhere in the code of `linuxwacom/xf86-input-wacom` but I'm no C programmer)

jigpu, 2025-01-08:
Data has been merged. This device doesn't seem to have a Wacom sensor, so you may want to file an issue with libwacom for assistance getting your device supported.

As for the tilt issue, you should reach out to the libinput project for assistance. It is supposed to be the ideal driver for non-Wacom devices. They should be happy to figure out what is going wrong.

commit f3684ae9 (2025-01-08): Add multiple sysinfo logs

* Lenovo Yoga 6 13ALC6
  sysinfo.mSCxAg3Y7r
  Closes #244

* Asus Q405UA-BI57T
  sysinfo.ROb2lagkqG
  Closes #243

* XP-Pen Deco L
  sysinfo.y9iVOhZiPM
  Closes #242

* Chuwi MiniBook X
  sysinfo.CSujtSF02A
  Closes #241

* GPD Pocket 3
  sysinfo.4p9auNwjkC
  Closes #240

* Dell Latitude 7210 2 in 1
  sysinfo.oN0r97Rlcn
  Closes #239

* XP-Pen Artist 12
  sysinfo.BPdqfy3fcP
  Closes #238

* Huion Kamvas Pro 13 (2.5K)
  sysinfo.m8z8elNbuM
  Closes #237

* Lenovo IdeaPad Flex 5 16ALC7
  sysinfo.ZBg72Fo2lq
  Closes #236

* Lenovo 20LHS04L00 ThinkPad X380 Yoga
  sysinfo. [... cut]