linuxwacom/wacom-hid-descriptors #247: Huion Kamvas 13 (GS1331)

theFroh, 2022-08-26:
Hey, this is showing up as an h950p and doesn't seem to let me calibrate for screen/pen parallax, nor recognise pen tilt. I'm also seeing:
_missing tablet capabilities: resolution. Ignoring this device_

**Device Name:** Huion Kamvas 13 (not the Pro variant)
**Model number:** GS1331

Not sure what steps to take other than submitting this sysinfo -- the button layout and binding seemed to work perfectly fine with it misidentified as an H950P.

[sysinfo.TkuyAF25Kf.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/9436342/sysinfo.TkuyAF25Kf.tar.gz)

blzq, 2022-12-04:
Generated a sysinfo under Xorg:

[sysinfo.Xzxg7g1kCc.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/10147836/sysinfo.Xzxg7g1kCc.tar.gz)

closed 2025-01-07 by jigpu

jigpu, 2025-01-07:
Data has been merged.

commit a0762983 (2025-01-07): Add multiple sysinfo logs

* GPD Pocket 3 / GXTP7380:00 #2
  sysinfo.zBQtvbeIj3
  Closes #262

* HP Envy x360 Convertible 13-ay0xxx
  sysinfo.9KfZlJn3lC
  Closes #261

* Data for Adding Huion Q11K
  sysinfo.5pgmqeQXK6
  Closes #259

* Lenovo Yoga 9i 15 [Wacom HID 5201]
  sysinfo.xbtOhaOv4r
  Closes #258

* Lenovo X12 Detachable
  sysinfo.3dthwuOp6q
  Closes #254

* Lenovo Yoga 7 16IAP7
  sysinfo.ilhzyGlYp4
  Closes #253

* HP ENVY x360 15-ee0000na
  sysinfo.YhyDAyKftq
  Closes #250

* UGEE EX07
  sysinfo.2uu4YqpezM.ugee-ex07
  Closes #249

* Huion Kamvas 13 (GS1331)
  sysinfo.TkuyAF25Kf
  Clos [... cut]