linuxwacom/wacom-hid-descriptors #252: Dell 5190 2in1

nagaisa, 2022-09-01:
Works with a non Wacom pen i have, tho pressure is inverted. Wacom pen gives nothing
[sysinfo.VBqR9LHvjY.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/9466653/sysinfo.VBqR9LHvjY.tar.gz)

jigpu, 2022-09-01:
Thanks for the sysinfo -- I've merged it and opened a pull request for libwacom to add a data file for this device.

As for the inverted pen pressure, since this device uses Wacom's alternate vendor ID (2d1f) the issue is going to be in one of the generic input drivers or caused by an incompatibility with the non-Wacom pen. To debug this, you'll need to install the "libinput-utils" and "hid-replay" packages from your package manager.

Please first use the following command to check how the userspace driver responds to changes in pressure. It should print out data whenever the pen is touching the device. Check to see if the "pressure" value increases/decreases with increasing/decreasing pressure. Press CTRL+C to stop the command.
~~~
$ sudo stdbuf -o0 libinput debug-events | awk '/pressure/'
~~~

If the pressure issue is present in the output of the previous command, use the following command to check how the kernel driver responds to changes in pressure. It will ask for a "device event number" -- type in the number associated with the "stylus" or "pen" device (e.g. I might see `/dev/input/event24:	Wacom Intuos Pro M Pen` in the list, so I would type in `24`). Like the last command, it should print out data whenever the pen is touching the device. Check to see if the values change as expected. Press CTRL+C to stop.
~~~
$ sudo stdbuf -o0 libinput record | awk '/PRESSURE/ { print $7 }'
~~~

If the pressure issue is *still* present, it may be an issue of your non-Wacom stylus not being compatible with your device. Try running the following command to dump the raw events sent by the hardware. It will ask you to choose a device: select the device with Wacom, pen, or stylus in its name. It should generate output whenever the pen is in proximity (if it doesn't, quit recording and try again with a different device number). Touch the pen down and then slowly increase and then decrease pressure. Remove the pen, and press CTRL+C to stop recording. You'll need to attach the generated `/tmp/isdv4-2d1f-002c.txt` file to this bug for review.
~~~
$ sudo hid-recorder | tee /tmp/isdv4-2d1f-002c.txt
~~~

nagaisa, 2022-09-01:
> Thanks for the sysinfo -- I've merged it and opened a pull request for libwacom to add a data file for this device.
>
> As for the inverted pen pressure, since this device uses Wacom's alternate vendor [...]

heres the file [isdv4-2d1f-002c.txt](https://github.com/linuxwacom/wacom-hid-descriptors/files/9471570/isdv4-2d1f-002c.txt) the pen does seam to have the inverted issue with every command. aslo is there something i can do to try and get the other pen(actually a Wacom pen) i have to work?

jigpu, 2022-09-01:
Hmm, yeah, its definitely inverted here as well. The status ("Sta") bits also appear to be incorrect; its showing the tip and side-switch to always be held down, even when (presumably) hovering and not pressing the switch.

~~~
               Len   ID    Sta   X        Y        Pres   ??   TiltX   TiltY
E: 0.000000    13    02    23    4c 44    43 0b    ff 0f   20   6c 07   4c 04
E: 0.000628    13    02    23    4c 44    43 0b    ff 0f   22   6c 07   4c 04
E: 0.001259    13    02    23    4c 44    43 0b    ff 0f   25   6c 07   4c 04
E: 0.002306    13    02    23    4c 44    43 0b    ff 0f   28   6c 07   4c 04
E: 0.008203    13    02    23    4c 44    43 0b    ff 0f   2a   6c 07   e8 03
E: 0.011776    13    02    23    4c 44    43 0b    ff 0f   2a   6c 07   e8 03
E: 0.018487    13    02    23    4c 44    43 0b    ff 0f   2a   6c 07   4c 04
E: 0.021952    13    02    23    43 44    41 0b    ff 0f   2b   6c 07   4c 04
E: 0.028313    13    02    23    43 44    41 0b    ff 0f   2b   08 07   4c 04
E: 0.031887    13    02    23    43 44    41 0b    ff 0f   2c   08 07   4c 04
[...]
E: 0.601949    13    02    23    0a 45    29 0b    dc 07   04   a4 06   84 03
E: 0.608260    13    02    23    0a 45    29 0b    3a 07   01   a4 06   84 03
E: 0.611903    13    02    23    0a 45    29 0b    96 06   01   a4 06   84 03
E: 0.618381    13    02    23    0a 45    29 0b    f5 05   00   a4 06   84 03
E: 0.621890    13    02    23    0a 45    29 0b    62 05   00   a4 06   84 03
E: 0.628349    13    02    23    0a 45    29 0b    f3 04   00   a4 06   84 03
E: 0.631879    13    02    23    0a 45    29 0b    a2 04   00   a4 06   84 03
E: 0.638148    13    02    23    0a 45    29 0b    6e 04   00   a4 06   84 03
E: 0.641647    13    02    23    0a 45    29 0b    58 04   00   a4 06   84 03
~~~

> is there something i can do to try and get the other pen(actually a Wacom pen) i have to work?

You can try running hid-recorder again and seeing if any events are generated in response to the Wacom pen. I have a hunch that nothing will happen though. Not all combinations of pen and sensor are compatible, even they're both from Wacom. The tech that lets the sensor wirelessly read pressure and other data has gone though several revisions and they can't always work with each other.

I believe [this](https://www.dell.com/en-us/shop/dell-active-emr-pen/apd/444-bbcs/handhelds-tablet-pcs) might be the "official" pen for the 5190. I see a lot of negative reviews, even from 5190 users though. Not sure why the pen would fail to work... An alternative might be the [Noris digital pen](https://www.staedtler.com/intl/en/discover/noris-digital) from Staedtler. The 5190 and tablets based on the same tech is mentioned in its compatibility list, and the pen has much better reviews on Amazon.

closed 2025-01-09 by jigpu