linuxwacom/wacom-hid-descriptors #259: Data for Adding Huion Q11K

cybereality, 2022-09-13:
HUION Inspiroy Q11K Wireless Graphic Drawing Tablet:

[sysinfo.5pgmqeQXK6.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/9561458/sysinfo.5pgmqeQXK6.tar.gz)

This tablet works fine in apps like GIMP, but doesn't seem to be recognized in the Ubuntu 22.04 settings menu, and certain other apps. I'd appreciate adding official support. Thanks.

https://www.amazon.com/gp/product/B071JN86L3/

jigpu, 2025-01-07:
Data has been merged. If this device still isn't recognized in the settings menu, you may want to reach out to libwacom for assistance getting your device supported.

closed 2025-01-07 by jigpu

cybereality, 2025-01-08:
Thanks a bunch!!! I'm actually using another tablet now, but that's still a popular model so it's appreciated.

commit a0762983 (2025-01-07): Add multiple sysinfo logs

* GPD Pocket 3 / GXTP7380:00 #2
  sysinfo.zBQtvbeIj3
  Closes #262

* HP Envy x360 Convertible 13-ay0xxx
  sysinfo.9KfZlJn3lC
  Closes #261

* Data for Adding Huion Q11K
  sysinfo.5pgmqeQXK6
  Closes #259

* Lenovo Yoga 9i 15 [Wacom HID 5201]
  sysinfo.xbtOhaOv4r
  Closes #258

* Lenovo X12 Detachable
  sysinfo.3dthwuOp6q
  Closes #254

* Lenovo Yoga 7 16IAP7
  sysinfo.ilhzyGlYp4
  Closes #253

* HP ENVY x360 15-ee0000na
  sysinfo.YhyDAyKftq
  Closes #250

* UGEE EX07
  sysinfo.2uu4YqpezM.ugee-ex07
  Closes #249

* Huion Kamvas 13 (GS1331)
  sysinfo.TkuyAF25Kf
  Clos [... cut]