linuxwacom/wacom-hid-descriptors #262: GPD Pocket 3 / GXTP7380:00 #2

michimussato, 2022-10-02:
[sysinfo.zBQtvbeIj3.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/9693295/sysinfo.zBQtvbeIj3.tar.gz)

michimussato, 2022-10-02:
```
[michael@pocket3 scripts] (master)$ ./git-update.sh /tmp/sysinfo.zBQtvbeIj3.tar.gz --url="https://github.com/linuxwacom/wacom-hid-descriptors/issues/262" --oem=GPD --product="Pocket 3 (Intel Core i7-1195G7)"
Manufacturer: "GPD"
Product: "Pocket 3 (Intel Core i7-1195G7)"
Is this correct? (Y/N) y
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  141k    0  141k    0     0   219k      0 --:--:-- --:--:-- --:--:--  219k
NOTE: Pen device is not a Wacom sensor. Tablet definition cannot be auto-generated.
NOTE: Unable to parse HID data. Skipping creation of libwacom tablet definition.
```

jigpu, 2025-01-07:
Data has been merged. As mentioned in your log output, this device doesn't seem to have a Wacom sensor, so you may want to file an issue with libwacom for assistance getting your device supported.

commit a0762983 (2025-01-07): Add multiple sysinfo logs

* GPD Pocket 3 / GXTP7380:00 #2
  sysinfo.zBQtvbeIj3
  Closes #262

* HP Envy x360 Convertible 13-ay0xxx
  sysinfo.9KfZlJn3lC
  Closes #261

* Data for Adding Huion Q11K
  sysinfo.5pgmqeQXK6
  Closes #259

* Lenovo Yoga 9i 15 [Wacom HID 5201]
  sysinfo.xbtOhaOv4r
  Closes #258

* Lenovo X12 Detachable
  sysinfo.3dthwuOp6q
  Closes #254

* Lenovo Yoga 7 16IAP7
  sysinfo.ilhzyGlYp4
  Closes #253

* HP ENVY x360 15-ee0000na
  sysinfo.YhyDAyKftq
  Closes #250

* UGEE EX07
  sysinfo.2uu4YqpezM.ugee-ex07
  Closes #249

* Huion Kamvas 13 (GS1331)
  sysinfo.TkuyAF25Kf
  Clos [... cut]
closed 2025-01-07 by jigpu