linuxwacom/wacom-hid-descriptors #269: ThinkPad X13 Yoga Gen 3

nilskanning, 2022-11-04:
The Wacom device in my ThinkPad X13 Yoga Gen 3 is currently not detected by `libwacom-list-local-devices` and the Gnome Wacom Tablet settings, see [sysinfo.2ZrtP6Q9a9.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/9941987/sysinfo.2ZrtP6Q9a9.tar.gz). It would be great, if you could add it. Thanks a lot!

spxak1, 2024-01-01:
Changing 5285 to 5284 in the .tablet for the Gen2 makes it identified. Also on Gnome Settings.

Calibration doesn't work though. 

Here's my .tablet file:
~~~
# Lenovo ThinkPad X13 Yoga Gen 3
# Sensor Type: AES
# Features: Touch (Integrated), Tilt
# HW Resolution: 28604 x 17878 (2540 x 2540 lpi)
#
# Autogenerated from sysinfo.AP2hbJxj8g
# https://github.com/linuxwacom/wacom-hid-descriptors/issues/188#issue-10964235>

[Device]
Name=ISDV 5284
ModelName=WACF2200
DeviceMatch=i2c:056a:5284
Class=ISDV4
Width=11
Height=7
IntegratedIn=Display;System
Styli=@isdv4-aes;

[Features]
Stylus=true
Touch=true
~~~

Save it as ```isdv4-5284.tablet``` in ```/usr/share/libwacom``` and reboot.

nilskanning, 2024-01-01:
I am experiencing the same behavior with that `isdv4-5284.tablet` file: The tablet is identified and works, except for calibration.

In addition, `libwacom-list-local-devices` yields `(libwacom-list-local-devices:8534): libwacom-WARNING **: Tablet 'i2c:056a:5284' has no buttons defined, do something!` which can be avoided by adding `Buttons=0` at the end of the config file, as is the case for `isdv4-5285.tablet` on my system.

closed 2024-12-23 by jigpu

jigpu, 2024-12-23:
Data has been imported into the repository. I've opened a pull request with the libwacom project to add a tablet definition file for this device.

commit 3bf58ad9 (2024-12-23): Add multiple sysinfo logs

* ASUS ROG Flow X16 GV601R
  sysinfo.layqUCS31O
  Closes #287

* Lenovo ThinkPad L13 Yoga Gen 3
  sysinfo.Oc1coPFLjt
  Closes #286

* Thinkpad x12 detachable
  sysinfo.hnYS6Pa8Vq
  Closes #285

* Huion 580X
  sysinfo.MafuzfBnxU
  Closes #284

* Lenovo Thinkpad X1 Yoga 1st: New variant
  sysinfo.8StWwkc2tP
  Closes #283

* Dell Inspiron 16 7620 2-in-1
  sysinfo.IZFFLfNznv
  Closes #282

* 2021 Eve V
  sysinfo.4B1hHLbz7d
  Closes #281

* Lenovo Ideapad Duet 5i
  sysinfo.vcgNVDx9C0
  Closes #278

* Lenovo Thinkpad X13 Yoga Gen2
  sysinfo.705kBQn3Nr
  Closes #277

* XP-P [... cut]