linuxwacom/wacom-hid-descriptors #271: Microsoft Surface 3

maxso-216, 2022-11-10:
This is surface 3 (non-pro) using Linux-surface kernel.
[sysinfo.cLx1gktLDV.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/9977862/sysinfo.cLx1gktLDV.tar.gz)
The touchscreen is working and stylish is recognized.

maxso-216, 2022-11-10:
However, when trying to use the  `git-update.sh`,it failed to gen and throw `NOTE: Pen device is not a Wacom sensor. Tablet definition cannot be auto-generated.` . It's seem it required handcraft generate.

maxso-216, 2022-11-11:
Also, I find out the tarball is containing information referring to the USB type cover keyboard, but not the touchscreen itself. I think there's something wrong with `sysinfo.sh` generation, as the touchscreen is using SPI (?) protocol.

closed 2024-12-23 by jigpu

jigpu, 2024-12-23:
Data has been imported into the repository. It seems like we may need to update the sysinfo script, as you mentioned. If you still have access to this device and are willing to help, let me know and I can re-open this issue to help you get a more complete dump!

Thertez, 2025-01-05:
Hello,

Does data being imported to the repo mean .tablet data file will be created for Surface 3? I am a bit confused on how everything works.
I am not the OP, but if anything else is needed to be provided I can try and help.

commit 3bf58ad9 (2024-12-23): Add multiple sysinfo logs

* ASUS ROG Flow X16 GV601R
  sysinfo.layqUCS31O
  Closes #287

* Lenovo ThinkPad L13 Yoga Gen 3
  sysinfo.Oc1coPFLjt
  Closes #286

* Thinkpad x12 detachable
  sysinfo.hnYS6Pa8Vq
  Closes #285

* Huion 580X
  sysinfo.MafuzfBnxU
  Closes #284

* Lenovo Thinkpad X1 Yoga 1st: New variant
  sysinfo.8StWwkc2tP
  Closes #283

* Dell Inspiron 16 7620 2-in-1
  sysinfo.IZFFLfNznv
  Closes #282

* 2021 Eve V
  sysinfo.4B1hHLbz7d
  Closes #281

* Lenovo Ideapad Duet 5i
  sysinfo.vcgNVDx9C0
  Closes #278

* Lenovo Thinkpad X13 Yoga Gen2
  sysinfo.705kBQn3Nr
  Closes #277

* XP-P [... cut]