linuxwacom/wacom-hid-descriptors #275: Lenovo Yoga 6 13ALC7

tealeg, 2022-12-04:
Note that is distinct from the Yoga 6 13ALC6 (there's a different issue for that).

I've already created the the libwacom tablet file so you'll see that in the sysinfo - also note that I'm on a very current, pure Wayland setup so there's no xinput output and no lsusb output.  

Anyhow, I'll submit a PR over on libwacom anyway.  If you you need anything more from me (I could easily live boot on a more *normal* distro and runs sysinfo again), just shoult.

tealeg, 2022-12-04:
[sysinfo.4gtjdCXyFN.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/10148332/sysinfo.4gtjdCXyFN.tar.gz)
Forgot to attach the file!

closed 2024-12-23 by jigpu

jigpu, 2024-12-23:
Data has been imported into the repository.

commit 3bf58ad9 (2024-12-23): Add multiple sysinfo logs

* ASUS ROG Flow X16 GV601R
  sysinfo.layqUCS31O
  Closes #287

* Lenovo ThinkPad L13 Yoga Gen 3
  sysinfo.Oc1coPFLjt
  Closes #286

* Thinkpad x12 detachable
  sysinfo.hnYS6Pa8Vq
  Closes #285

* Huion 580X
  sysinfo.MafuzfBnxU
  Closes #284

* Lenovo Thinkpad X1 Yoga 1st: New variant
  sysinfo.8StWwkc2tP
  Closes #283

* Dell Inspiron 16 7620 2-in-1
  sysinfo.IZFFLfNznv
  Closes #282

* 2021 Eve V
  sysinfo.4B1hHLbz7d
  Closes #281

* Lenovo Ideapad Duet 5i
  sysinfo.vcgNVDx9C0
  Closes #278

* Lenovo Thinkpad X13 Yoga Gen2
  sysinfo.705kBQn3Nr
  Closes #277

* XP-P [... cut]