linuxwacom/wacom-hid-descriptors #279: Lenovo Yoga 7 Gen 7 14ARB7

stuarthayhurst, 2022-12-24:
[sysinfo.vZi1KLwSip.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/10299063/sysinfo.vZi1KLwSip.tar.gz)

imsofi, 2022-12-25:
Further sysinfo dump, based on Fedora 37, GNOME Xorg.

[sysinfo.tAAUZfAlhq.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/10300479/sysinfo.tAAUZfAlhq.tar.gz)

imsofi, 2023-03-29:
@stuarthayhurst Can you verify which screen your computer is based on following [PSREF](https://psref.lenovo.com/Product/Yoga/Yoga_7_14ARB7)?

My sysinfo dump is based on:

`14" 2.8K (2880x1800) OLED 400nits Glossy, 100% DCI-P3, 90Hz, DisplayHDR True Black 500, Dolby Vision, TÜV Low Blue Light, Glass, Touch`

And shows up under `HID 52D3` instead of `HID 52D5`. So the digitizes may be different between the display options.

stuarthayhurst, 2023-03-29:
I have the 2.2K screen model, so that's probably it then. If you want to add your files to #280, I can add you as a contributor, or you can probably submit a PR to my fork, whatever you prefer.

jigpu, 2024-12-10:
Thanks to both of you for your reports. They have both been merged.

closed 2024-12-10 by jigpu