linuxwacom/wacom-hid-descriptors #294: HP Envy x360 Convertible 13-ar0xxx

raymond-nee, 2023-02-18:
Dump from script: [sysinfo.tqYLZBiOyb.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/10774156/sysinfo.tqYLZBiOyb.tar.gz)

Tried adding the id ```04f3:23b9```  to the tablet files in ```/usr/share/libwacom``` and KDE Plasma can recognize the screen and stylus (HP Tilt Pen) . Screen touch, pressure sensetivity and stylus button are work.

The stylus button remap only works under X11, and the system warns the screen "is going to run out of battery".

7bjvnj2, 2023-08-10:
@raymond-nee  hi can you please explain me  how to do the whole process from the scratch

raymond-nee, 2023-08-12:
@7bjvnj2 
For collecting the sysinfo, just simply run the following command under a **X11 session**.
`wget -qO- https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh | sudo sh` 

For the addition of tablet ID, I just followed all the steps listed in [libwacom Readme](https://github.com/linuxwacom/libwacom#adding-tablet-descriptions-to-libwacom).

7bjvnj2, 2023-08-24:
@raymond-nee  thanks but i still didn't get it is there any video walk-through i can follow ?

closed 2024-12-21 by jigpu

jigpu, 2024-12-21:
Data has been imported into the repository. You may need to work with the libwacom project to create a tablet definition file for this device.

commit 7d1fb808 (2024-12-21): Add multiple sysinfo logs

* Lenovo Yoga C930 (again)
  suysinfo.ZX9zSC40wj
  Closes #312

* Dell Latitude 5300 2-in-1
  suysinfo.EbblcgnWr0
  Closes #311

* Huion Inspiroy RTM-500
  suysinfo.X8pBxWhnQo
  Closes #310

* Asus Zenbook UX363J
  suysinfo.KQnersNEFa
  Closes #309

* Lenovo Chromebook Duet 5
  suysinfo.bC7535dy9c
  Closes #308

* Dospara Raytrektab 8inch RT08WT
  suysinfo.G68KvjOZcA
  Closes #305

* Samsung Galaxy Book3 Pro 360 (NP960QFG-KA1US)
  suysinfo.0cYJGV5SS0
  Closes #304

* XP-Pen Deco Pro SW
  suysinfo.YgZdvbyG6s
  Closes #300

* XP-Pen artist 12 Pro
  suysinfo.QpUbKzqb5v
 [... cut]