linuxwacom/wacom-hid-descriptors #296: Lenovo ThinkPad X1 Yoga 1st

jayaddison, 2023-03-07:
[sysinfo.cKbD6GCsrw.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/10911925/sysinfo.cKbD6GCsrw.tar.gz)

jayaddison, 2023-03-07:
Note: the settings from the similar [`ISDv4 5093`](https://github.com/linuxwacom/libwacom/pull/492) (a variant of the same laptop model, perhaps) seem to work well with this tablet and stylus, after updating the `DeviceMatch` to include entries for both.

(but perhaps it should exist as a separate `.tablet` file anyway)

closed 2024-12-21 by jigpu

jigpu, 2024-12-21:
Data has been imported into the repository. I've opened a pull request with the libwacom project to add a tablet definition file for this device.

jayaddison, 2024-12-21:
Thank you, @jigpu!

commit 7d1fb808 (2024-12-21): Add multiple sysinfo logs

* Lenovo Yoga C930 (again)
  suysinfo.ZX9zSC40wj
  Closes #312

* Dell Latitude 5300 2-in-1
  suysinfo.EbblcgnWr0
  Closes #311

* Huion Inspiroy RTM-500
  suysinfo.X8pBxWhnQo
  Closes #310

* Asus Zenbook UX363J
  suysinfo.KQnersNEFa
  Closes #309

* Lenovo Chromebook Duet 5
  suysinfo.bC7535dy9c
  Closes #308

* Dospara Raytrektab 8inch RT08WT
  suysinfo.G68KvjOZcA
  Closes #305

* Samsung Galaxy Book3 Pro 360 (NP960QFG-KA1US)
  suysinfo.0cYJGV5SS0
  Closes #304

* XP-Pen Deco Pro SW
  suysinfo.YgZdvbyG6s
  Closes #300

* XP-Pen artist 12 Pro
  suysinfo.QpUbKzqb5v
 [... cut]