linuxwacom/wacom-hid-descriptors #339: HP Spectre x360 Convertible 14-ef2xxx

lalisand, 2023-11-23:
[sysinfo.za9NEmybk7.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/13452908/sysinfo.za9NEmybk7.tar.gz)

closed 2024-12-20 by jigpu

jigpu, 2024-12-20:
Data has been imported into the repository. You may need to work with the libwacom project to create a tablet definition file for this device.

lalisand, 2024-12-30:
Thank you. In fact, I have been using tablet definition file "ELAN-2513.tablet" with no apparent issues. This convertible uses the same display as many other HP Spectre devices with 14 inch display at 3:2 aspect ratio.

If you think, however, it would be better to create a dedicated tablet definition file, I am open to join the effort.

lalisand, 2025-01-25:
@jigpu since adding the data into the repository, my tablet stopped working and I have no idea how to fix it. As I mentioned, it worked before with the "ELAN-2513.tablet" (in fact even without it  - the only issue was wacom settings unavailable).

lalisand, 2025-03-16:
@jigpu the problem still persists, could you please advise on how to resolve this?

commit e765b5d4 (2024-12-20): Add multiple sysinfo logs

* Gotop T501
  sysinfo.HRXMv9isCi
  Closes #348

* Lenovo Yoga Book 9i (13IRU8)
  sysinfo.zpUboT0IUG
  Closes #346

* Gaomon 1080Pro
  sysinfo.gYsbNOKDja
  Closes #344

* Lenovo yoga c940 (14")
  sysinfo.PyrukvEvUN
  Closes #340

* HP Spectre x360 Convertible 14-ef2xxx
  sysinfo.za9NEmybk7
  Closes #339

* Data for isdv-48b7
  sysinfo.XkevsUVUax
  Closes #338

* XP-Pen G430s
  sysinfo.slpdTWkTPl
  Closes #337

* LENOVO Yoga 7 14ARP8
  sysinfo.cIIJCRa5QX
  Closes #336

* Thinkpad X1 tablet gen1
  sysinfo.zxVmXyoORE
  Closes #335

* XP-Pen Artist 13.3 Pro
  sysinfo.WaL [... cut]