linuxwacom/wacom-hid-descriptors #365: Espresso 15 Portable Display

davepwsmith, 2024-03-04:
[sysinfo.pzZajuS1x3.tar.gz](https://github.com/linuxwacom/wacom-hid-descriptors/files/14480014/sysinfo.pzZajuS1x3.tar.gz)

Attached is the dump for the [Espresso 15 display](https://au.espres.so/products/espressodisplays-and-stand).

Seems to work OK on wayland with a customized `.tablet` entry in /etc/libwacom/ but the touch outputs to the main laptop display whilst the stylus outputs to the external espresso 15 display. Not sure how to remedy that, or if this helps!

davepwsmith, 2024-03-05:
If anyone is looking for how to get this display working properly in Gnome/Wayland, I've put some [instructions here](https://www.reddit.com/r/EspressoDisplays/comments/1b72fhf/linux_with_gnomewayland_setup_script_for_espresso/).

closed 2024-12-16 by jigpu

jigpu, 2024-12-16:
I've created a pull request to add this device's sensor information to libwacom. You shouldn't need to do anything else for this device to be supported by libwacom in the near future.