linuxwacom/wacom-hid-descriptors #366: Dell latitude 7320 Detachable

nos1609, 2024-03-16:
Looks like [the IDs](https://github.com/linuxwacom/wacom-hid-descriptors/issues/174) need some refreshing:

```
[    2.095024] input: WCOM49A5:00 056A:49A5 Touchscreen as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM49A5:00/0018:056A:49A5.0002/input/input10
[    2.095078] input: WCOM49A5:00 056A:49A5 as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM49A5:00/0018:056A:49A5.0002/input/input11
[    2.095101] input: WCOM49A5:00 056A:49A5 Stylus as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM49A5:00/0018:056A:49A5.0002/input/input12
[    2.095135] input: WCOM49A5:00 056A:49A5 as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM49A5:00/0018:056A:49A5.0002/input/input13
[    2.095168] input: WCOM49A5:00 056A:49A5 Mouse as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM49A5:00/0018:056A:49A5.0002/input/input14
[    2.095215] hid-generic 0018:056A:49A5.0002: input,hidraw1: I2C HID v1.00 Mouse [WCOM49A5:00 056A:49A5] on i2c-WCOM49A5:00
[    2.234480] wacom 0018:056A:49A5.0002: hidraw1: I2C HID v1.00 Mouse [WCOM49A5:00 056A:49A5] on i2c-WCOM49A5:00
[    2.234534] input: Wacom HID 49A5 Pen as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM49A5:00/0018:056A:49A5.0002/input/input15
[    2.234852] input: Wacom HID 49A5 Finger as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM49A5:00/0018:056A:49A5.0002/input/input16
```

nos1609, 2024-03-16:
Also found in https://github.com/linuxwacom/wacom-hid-descriptors/issues/347

nos1609, 2024-03-16:
Initial work done in https://github.com/linuxwacom/wacom-hid-descriptors/issues/174

closed 2024-12-16 by jigpu

jigpu, 2024-12-16:
I've created a pull request to add this device's sensor information to libwacom. You shouldn't need to do anything else for this device to be supported by libwacom in the near future.