linuxwacom/wacom-hid-descriptors #448: HP Envy x360 2-in-1 Laptop 14-fc0xxx

lukee12zakk, 2025-01-29:
I'm submitting a sysinfo file as my laptop does not have pen support and my pen is just seen as a mouse. 

[sysinfo.NjygX8Ofze.tar.gz](https://github.com/user-attachments/files/18585969/sysinfo.NjygX8Ofze.tar.gz)

jigpu, 2025-01-29:
Thanks for the sysinfo. I've added this data to the repository. It looks like this device uses an Elan sensor, so I would suggest working with the DIGImend developers (https://github.com/DIGImend/digimend-kernel-drivers) to figure out why the pen does not work. You can also work with the libwacom maintainer (https://github.com/linuxwacom/libwacom) to create a tablet definition file to let tools like GNOME Control Center configure the pen after the driver works properly.

closed 2025-01-29 by jigpu

lukee12zakk, 2025-01-29:
Hello.

I was just asking how would i get get the tablet file from libwacom. I checked and they have my pen/screen model but not for the hp laptop that i have. 

I'm new to Linux, GitHub and those things so its a bit hard for me.

Thanks

commit 34f18591 (2025-01-29): Add HP Envy x360 2-in-1 Laptop 14-fc0xxx

* HP Envy x360 2-in-1 Laptop 14-fc0xxx
  sysinfo.NjygX8Ofze
  Closes #448